mgenelins/w0yc-repeaters #4: Replace 462.650 commercial antenna

mgenelins, 2021-06-08:
The Metro 4 repeater antenna has died back in 2019, but it was not replaced. Order a new antenna (perhaps 
VHF/UHF dual bander, low maintenance, broadband) to replace this antenna on St. Paul Campus.

mgenelins, 2023-03-29:
Metro 4 seems to be getting deafer in March 2023. After returning from my Orlando, FL vacation, my portable no longer holds Metro 4 solidly.

+1 on the prioritization for this.

mgenelins, 2023-09-25:
Ran NanoVNA and Watt-meter on the feedline for VHF/UHF Tram antenna. Found 5:1 SWR.
- [ ] Replace antenna

mgenelins, 2023-10-13:
- [ ] Test out the existing UHF Commscope DB420 antenna located in Matt's Garage for use on 462MHz.

mgenelins, 2024-06-01:
Ordered replacement antenna from amazon:

HYS Pre-Tuned 462MHz-467MHz UHF Amateur Base Antenna, GMRS Base Station Antenna, Outdoor 7.5ft (86.6inchs) Long-Range 8.5dBi, Pole Mast Mount (SO239-UHF Female)

https://www.amazon.com/dp/B0CRTRQYKY?psc=1&ref=ppx_yo2ov_dt_b_product_details

installed june 1 2024.

mgenelins, 2024-06-01:
SWR is now 1.32:1. 13 watts out, so need to replace the transmitter next.

mgenelins, 2024-06-01:
New west antenna is donatred by John KA0KMJ and connected to 442.425 Quantar. Still unacceptable SWR reading, so bad connector probably?

KD0TLS, 2024-06-01:
Not to pile on, but I can't bring up Metro 2 anymore. It transmits an ID,
but isn't receiving.
Even weirder, I can tickle Metro 4 with the same setup that fails to work
for Metro 2.

Todd Dugdale
 KD0TLS/KAF6430
 Plymouth, MN

mgenelins, 2024-06-02:
@KD0TLS Sounds like a fresh issue for Metro 2.

mgenelins, 2024-06-02:
Moved antennas around this afternoon again; need to re-test the SWR on the new Metro 4 antenna. Seems to have less coverage again, hard to bring up from Highland Park.

mgenelins, 2024-06-02:
Greetings Club members,

A HUGE thank you to those who showed up to help with work at St. Paul Campus repeater site today. Lots of (warm) fun was had by all. Today we:

. Replaced two antennas with new antennas.
. Ran Ian's antenna analyser to find a fault on one of our feelings.
. Replaced the broken GMRS antenna on the WEST mount point with a 442Mhz antenna donated by John KA0KMJ.
. Replaced a working VHF/UHF Tram antenna with a dedicated 462 MHz GMRS repeater antenna on the NORTH mount.
. Found the feeling to be bad that feeds the WEST mount.
. Moved the Motorola Quantar 442.425 FM/P25 repeater over to the VHF/UHF dual bander on the SOUTH mount. Tested 1.7:1 SWR, super improved coverage.

While it's not the full list of stuff that needs doing, I also updated the various GitHub issues with the current status of things and am getting closer to closing some of the longer running issues. *Phew*

After a morning celebrating Jack Vaplon's graduation from High school, I plan to meet Ian mid-afternoon at the site for another swing at the TXDELAY setting issue (we tried, but didn't have the right baud or terminal cable.) I also want to verify that the GMRS repeater is connected to the NORTH GMRS antenna and validate the SWR reading. After a morning of improved coverage, the coverage of Metro 4 seems to have gone back down again, and a quick check of the SWR might help clarify which cable is which.

Weather and time permitting, we'll correct the cabling issues and antenna connections.

73's,
-matt 

-----
Regards,
---Matt Genelin---
Vice President 
[Gopher Amateur Radio Club at The University of Minnesota](https://w0yc.org/)
Sr. Solutions Architect, DevOps Engineer
ARS N0YNT/W0USD, GMRS WQCQ790
Cell: +1 (612) 605-5382 Texts or call.
NNNNN

-=-=-=-=-=-=-=-=-=-=-=-
[Groups.io](http://groups.io/) Links: You receive all messages sent to this group. View/Reply Online (#362): https://groups.io/g/w0yc/message/362 Mute This Topic: https://groups.io/mt/106438387/5431698 Group Owner: [w0yc+owner@groups.io](mailto:w0yc+owner@groups.io)
Unsubscribe: https://groups.io/g/w0yc/leave/11992812/5431698/397617542/xyzzy [[mgenelin@fastcomputerserviceco.com](mailto:mgenelin@fastcomputerserviceco.com)]
-=-=-=-=-=-=-=-=-=-=-=-

 
MattyG
mgenelin=fastcomputerserviceco.com@groups.io

mgenelins, 2024-06-04:
Metro 4 is performing better then before new antenna install, not to system specs.
- [ ] Notify mygmrs.com people that a new antenna has been installed and we are ready for coverage testing.
- [ ] Program family PL tone from Kevin request from two months ago.
- [ ] Notify Kevin that his family PL tone is activated.
- [ ] Bring watt meter and re-check SWR level on Metro 4.
- [ ] Run NanoVNA through duplexor on Metro 4. Was putting out 13 watts
- [ ] Repogram RF decks on Metro 4 Transmitter to higher power to make up for 13 watts previously measured.

closed 2024-06-04 by mgenelins