mgenelins/w0yc-repeaters #6: Ham Lake: Switch out 224.940 for 224.820 - New System Build

mgenelins, 2021-06-08:
Ralph N0AWN and Matt N0YNT have a new Bridgecomm 222 MHz repeater in stock to replace the old 224.940. Matt has verified June 2021 that a low-coverage, coordinated, 224.94 exists and is on the air in Rosemount, MN. It's time to move 224.94 out of the metro to achieve coordination.

mgenelins, 2021-12-02:
Scheduled for a site install on Saturday, December 4, 2021 at 1300 on-site in Ham Lake.

mgenelins, 2021-12-02:
- [ ] Still need to complete coordination paperwork for a new system. 
- [ ] Move Email thread to this issue to track coordination technical specifications.

KD0TLS, 2021-12-03:
Thanks for the update! Unrelated, but I hope this brings the Ham Lake digipeater closer to reality.  Todd D.From: Matt GenelinSent: Thursday, December 2, 2021 8:16 AMTo: mgenelins/w0yc-repeatersCc: SubscribedSubject: Re: [mgenelins/w0yc-repeaters] Ham Lake: Switch out 224.940 for 224.820 - New System Build (#6) [ ] Still need to complete coordination paperwork for a new system.[ ] Move Email thread to this issue to track coordination technical specifications.

mgenelins, 2021-12-08:
@KD0TLS Not sure about the Ham Lake Digipeater? Can you tell me more on who was thinking that might be a thing?

mgenelins, 2021-12-08:
Pending:
- [ ] Still need to complete coordination paperwork for a new system.
- [ ] Audio level adjustment on 224.820 to bring UP Rx levels, and DOWN the beeps and ID levels.
- [ ] Replace remote base with a Maxtrac that has proper audio and PL levels.

KD0TLS, 2021-12-09:
You were telling me Ralph was doing that.

mgenelins, 2021-12-09:
@KD0TLS Oh, okie. Not sure if I have any updates from Ralph on the digipeater topic. He hasn't mentioned anything memorable to me in awhile.

mgenelins, 2022-02-21:
Ham Lake site visit today at 2pm CT 2-20-2022:

Replaced remote base with another broken Motorola GM300 radio; RX bad on ebay radio. Put old remote base radio back into service, re-tuned audio and TX. TX is drifting off-frequency, which causes RX to drift. 

- [ ] Need to replace Remote base radio entirely.

Disabled 180 sec. timeout by re-programming the BridgeComm repeater itself. 
- [ ] 30 Watts is max power on this system; voice IDs should match that, not 50 Watts. Need to fix.

Turned down the beep levels with the "tone" pot inside the RC210 controller. Voice IDs are too loud still; Dan N0KFB suggested that we disable all voice IDs. 
- [ ] Matt to re-program out those Voice ID's and re-visit on future date.

Ralph N0AWN suggested that we upgrade to a Motorola CDM 1550 or CDM 1550. 
- [ ] Matt to acquire one from eBay and program and get ready for a future site visit.

RC210 V 7.611 controller does not seem to take full programming load. Spews an error when programming; likely software version mis-match. Controller likely needs a firmware upgrade to a version that we have software for. PC software is v 7.610, firmware on controller is v7.611.

- [ ] Matt to program up the old RC210 controller on V6.x.x as a hot spare in case this controller keeps giving us issues. Have on the ready for hot swap for Ralph.

mgenelins, 2023-03-29:
Okie, this issue is getting stale. I am going to move the action items to individual issues and close this one as stale.

closed 2023-03-29 by mgenelins