mgenelins/w0yc-repeaters #10: RedCross Repeater @ Moos: License expired

mgenelins, 2021-12-03:
The commercial License for the Colocated Minneapolis Red Cross Repeater at Moos has expired. Email attempts the last several months to get someone to make contact, dispatch or enter into any agreement have been unsuccessful.
[image]
[image]

mgenelins, 2021-12-03:
FCC ULS Database screen shot:

[image]

mgenelins, 2021-12-03:
Pic of repeater at Moos:
[image]

KD0TLS, 2021-12-05:
Wow, I’ve never really seen the famed Moos tower station. Does this licensing issue potentially leave an open slot at Moos? From: Matt GenelinSent: Friday, December 3, 2021 12:19 PMTo: mgenelins/w0yc-repeatersCc: SubscribedSubject: Re: [mgenelins/w0yc-repeaters] RedCross Repeater @ Moos: License expired [KSR994] (Issue #10) Pic of repeater at Moos:

mgenelins, 2021-12-06:
@KD0TLS Moos tower Unmasked! Ha! I forget that not everyone has visited the famed room 140 on the 19th floor. Yes, that's a sneak preview of what things look like. I should really post a full set of pictures of all the racks.

To answer the question of if a new antenna has opened up- not yet. The Minneapolis Red Cross insists that they are using this 'critical piece of infrastructure' yet nobody from the club has heard any traffic on their repeater. We've tried for months to setup a sked, to no availability.

So, I am advocating that we release this antenna from 453.xxx for other uses. I have been advocating via email with @Ben Franske that we simply repurpose the repeater as DMR and offer up a talk group to all non-profits on such a future system. It would make more sense, and make better use of the UHF commercial spectrum.

mgenelins, 2022-02-11:
More email back and forth. RedCross states that system is off-air, cannot test correctly. Dan Peitso tried to schedule a weekday test, @mgenelins asked for a weekend test, PL tone and frequency. Never heard back.

This system is pending email comms from:
- [ ] Someone from the RedCross that can provide an updated license.
- [ ] Ben Franske to email @mgenelins a real, original, contact person to see if we can decomm the system. 

It's likely that this system was installed before the written agreements that Ben had created for Memorandum of Understandings that the club now uses. This may be a political issue in the future.

mgenelins, 2022-10-17:
Ben emailed back in August 2022 Matt to approve decomissioning of this system and site removal.

System removed from Site during October 15 2022 visit. Sitting in Matt's garage.

closed 2022-10-17 by mgenelins