mgenelins/w0yc-repeaters #14: NXDN system move from STP campus and reprogram

Deathn00dle, 2022-10-09:
NXDN repeater from STP to WSP. Reprogram to new split. Duplexer tuning sesh at Matt casa to figure out NanoVNA.

mgenelins, 2022-10-11:
I think this would be an awesome move - the first NXDN repeater on the air in the Twin Cities that I am aware of!

Perhaps we should get an FM machine on the VHF pair to hold the channel from re-coordination. I am aware of two MSF5000's sitting at the W0YC ham shack, one that is programmed up and ready for a controller to be wired to it. Perhaps a split-site between St. Paul (Transmitter) and Minneapolis (receiver)? Let me add an issue for that!

mgenelins, 2023-03-29:
Tristen and Nate have volunteered at the end of April to replace 147.150 with a Motorola Quantar in mixed-mode setup. FM/P25 is the future for this repeater.

The Kenwood project can move over to 145.230 as SNP pair and run Kenwood NexEdge solely.

mgenelins, 2023-04-19:
@Deathn00dle Just double checking that I still have this issue open. I do - and I intend that we should upgrade 147.150 to a NXDN machine next.

We'll likely need to move this to another VHF repeater pair like 145.230 for awhile while we await coordination for a new VHF pair. 

The 147.150 repeater is scheduled for a mix-mode upgrade to a Motorola Quantar for NB FM and P25 capabilities later in April.

mgenelins, 2023-09-23:
I am calling this issue done, as system is at Jason's house for repgoramming and bruning in. 
A new issue has been created:
https://github.com/mgenelins/w0yc-repeaters/issues/30

To track progress on various items related to this new NextEdge repeater.

closed 2023-09-23 by mgenelins