mgenelins/w0yc-repeaters #23: 444.200 causing interference

Deathn00dle, 2023-07-06:
Causing interference with Hennepin Co. Apparently their pairs are on 444.20625 and 444.1375 or something. 

1) Take NH7CY/R off air 
2) Find and reprogram NH7CY/R to new working pair

mgenelins, 2023-07-06:
I have removed the K4USD network connection from NH7CY/R 444.200 CC 11 Minneapolis. This will stop the network traffic entirely from the system.

443.000 N0YNT/R on 118.8hz is available, and already coordinated to that site. See [the MRC repeater list](https://www.mrc.gen.mn.us/MN_List.pdf) to verify. We can move it at a future date to that SNP (Shared Non-Protected) pair in wait of a better pair.

mgenelins, 2023-07-06:
I have a visit scheduled for tomorrow at 8:00am.

mgenelins, 2023-07-06:
Communicated with the w0yc@groups.io email list:
From: Matt Genelin <mgenelin@gmail.com>
Reply-To: mgenelin@gmail.com
To:   mailing list <w0yc@groups.io>
Date: Wednesday, July 5 2023 at 10:21 PM CDT

Subject: Fwd: Re: [mgenelins/w0yc-repeaters] 444.200 causing interference (Issue #23)

Howdy W0YC Group,

Just an FYI: Jason and I received a complaint from the Hennepin Co. RACES group that the NH7CY/R on 444.200 CC 11 Minneapolis (Moos) is interfering with an unpublished set of two Hennepin Co. WC0HC/R DMR machines on splinter pairs around 444.200.

I have disabled the K4USD network connection from the system this evening. I have a trip planned tomorrow to the site to pull the power.

Jason and I are considering moving it to the 443.000 SNP pair that the portable repeater at the W0YC haystack currently occupies; we could run it in mixed mode (DMR and FM) with a tone of 118.8hz. That pair has been coordinated with Moos for several years, and the coordination is current.

You can follow along with Jason and my progress by watching GitHub issue #23, here:
https://github.com/mgenelins/w0yc-repeaters/issues/23 [1]

73's,
-matt

--
[1] https://github.com/mgenelins/w0yc-repeaters/issues/23

leoninelion, 2023-07-06:
Reminder to:
Have 6224P/6248P switch configured for Rack B
Connect router LACP feeds to core 6248P  
Single mode fiber interconnect if fiber/optics in hand.  

Not a priority, just a reminder for onsite activities.

mgenelins, 2023-07-06:
> @ github. com>

Good Morning, Lion,

Can you add those to the description on this GitHub issue, here:

https://github.com/mgenelins/w0yc-repeaters/issues/24

Warmly,

-matt

Regards,
---Matthew Genelin---
Fast Computer Service Co.
612.605.5382

mgenelins, 2023-09-20:
This system was moved to 443.000 CC11 a few weeks back. 
Recent visit I re-seated the connectors to the feedline, and it seems to be back on the air.

- [X] Matt to re-enable NH7CY/R on the cbridge.

closed 2023-09-20 by mgenelins