mgenelins/w0yc-repeaters #24: Todo List for Next Moos Site Visit

mgenelins, 2023-07-06:
Lion to fill in details here, please.

leoninelion, 2023-07-06:
To be done:
Dell 6224P switch to GFZ to be configured, must bring dual SFP+ 10GbE expansion card  
Install Dell 6224P in rack B in Moos
Install 10GbE optics in both 6224P (B) and 6248P (A)
Install singlemode dual fiber LC patch cable from Rack A to B, disconnect copper crosslink https://www.amazon.com/Fiber-Patch-Cable-Single-Duplex/dp/B00T89DIQW

Install 2x CAT5/6 from router OPT1/OPT2 to Rack A 6248P
Install POE splitter for RPi https://www.amazon.com/ANVISION-Splitter-Compliant-Raspberry-AV-PS05-1/dp/B07WCGY7KJ
Install POE splitter for echolink RPi (same purchase as above)

leoninelion, 2023-07-06:
ADD Remove old RPi power supplies once POE power is installed

mgenelins, 2023-10-13:
- [x] Order placed for 4 of the Ethernet PoE to USB/Ethernet adapters.
- [x] Order placed for 1 30 ft optical fiber cable.

@leoninelion Do all the ports on the switch have PoE enabled, or are there special ports that we should be using for PoE?
@leoninelion Does the existing Moos switch have the needed 10Gbe Optics installed already, or do we have these in sock or need ordering?

@leoninelion For "Dell 6224P switch to GFZ to be configured, must bring dual SFP+ 10GbE expansion card" how/what needs to be configured? Just slide a card in, or something more?

leoninelion, 2023-10-13:
@mgenelins All ports are POE enabled.
@mgenelins Optics need to be purchased:  https://www.amazon.com/10GBase-LR-Transceiver-Compatible-Ubiquiti-Mikrotik/dp/B0B56NWD7P

@mgenelins Turn the switch off, plug in the card to the outside slot, turn switch back on.