mgenelins/w0yc-repeaters #27: W0YC-5 APRS Node DOWN: Reported September 17th, 2023

mgenelins, 2023-09-20:
Todd, KD0TLS reports that the W0YC-5 APRS node is offline.

- [X] Matt has rebooted it twice, but the **_Xastir_** APRS software no longer running/starts at boot.

Checking the logs, 192.168.38.50 is running Ubuntu Wheesy, from 2014. So that's 9 years of service this equipment has gotten us.

- [ ] Likely time to take down and replace.

mgenelins, 2023-09-20:
The APRS gateway W0YC-5, sited in Moos Tower, has been off-air and with no connection to the -IS for six days now.

The station is the core gateway for the metro, due to its central location and wide coverage.

Both RF and internet functionality has been lost. Usually this indicates that the RPi needs a reset, though it could also point to power supply issues.

 

Todd D.

 KD0TLS

 Plymouth, MN

mgenelins, 2023-09-20:
En route to the site today to pick up the Rasp. pi, hat and radio. Back to matt's garage.

KD0TLS, 2023-09-20:
When I noticed W0YC-5 was down, I notified a few other operators that their gateways were down and they (mostly) got restored. N9MEC is a pernicious problem, but the metro coverage is doing fairly well without W0YC-5. Tomorrow is #APRSThursday, and that will be a stress test for the network w/o the big TX gateway. Overall, though, I’ve been surprised how the rest of the infrastructure is covering up the loss. Some DX will  be lost, some messaging will be lost, some low-power stations won’t get all of their packets gated, but it’s far from a disaster.  W0YC-5 has done an immense favor to the metro, so let’s take the time to replace it properly. I see no major urgency, but that’s just my opinion.  Todd D.  KD0TLS   From: Matt GenelinSent: Wednesday, September 20, 2023 11:36 AMTo: mgenelins/w0yc-repeatersCc: SubscribedSubject: Re: [mgenelins/w0yc-repeaters] W0YC-5 APRS Node DOWN: Reported September 17th, 2023 (Issue #27) En route to the site today to pick up the Rasp. pi, hat and radio. Back to matt's garage.

mgenelins, 2023-09-23:
Picked up radio and node, is at matt's garage now.

mgenelins, 2023-09-25:
Two possible routes:

- [ ] Ethan volunteered to see if he could get the old RaspPi-based APRS node software working again. Matt is to drop off the radio, PI, and TNC hat at Hamshack for Ethan to work on.

- [ ] Matt has ordered a replacement HUB for the Dell Virtual Machine to attempt the K1LEO Route of using a software-based TNC with USB cord attached to correct this faster. Amazon ordered the 10-port hub to be connected near VGA ports (the USBs near there) on the back of the virtualization machine. Ships to Matt in a couple of days.

mgenelins, 2023-10-13:
- [x] Orange USB KISS TNC has been located! Now to get a radio interfaced to it.
- [x] Lion has a radio ready to go for this TNC. 
- [ ] Matt to pick up radio from Lion and install TNC + radio at Moos.

KD0TLS, 2023-10-14:
A Google search didn't provide any info as to what an "Orange USB KISS
TNC" is. What software would be running? Will the APRS gateway be sharing
an antenna with Metro 2 again?
 I had hoped that a complete overhaul would provide additional
functionality that the previous node lacked: bulletins, groups, object
creation/maintenance, and remote access. Still, anything is better than
nothing. Maybe some of this could be provided by an -IS station running on
the server?

mgenelins, 2023-12-27:
Node has been replaced, Lion emailed Todd on the details of the modem and equipment stuffs. 
Closing this issue as fixed!

closed 2023-12-27 by mgenelins