mgenelins/w0yc-repeaters #29: Metro 2: Off air - Power Back on

mgenelins, 2023-09-20:
During Matt's Sept 20, 2023 Site visit, Matt powered off Metro 2 to safely move the antenna port from the APRS unit duplexor directly to Metro 2. The Normal config is Metro 2 + APRS unit shares a dual band antenna. Without the APRS VHF radio on the duplexor, an impediance mismatch would result, so the VHF/UHF duplexor has been removed and connected directly to the Metro 2 duplexor. 

Matt forgot to turn the repeater back on after the site visit.

- [ ] Press the power button on the front of the repeater to turn Metro 2 back on.

ethanleep, 2023-09-21:
Powered the repeater back on last night so this issue can probably be closed if it's responding as expected.

KD0TLS, 2023-09-21:
Metro 2 is back on the air! I've ker-chunked it a few times and heard it
ID.
Thanks, Ethan.

mgenelins, 2023-09-23:
@KD0TLS We moved Metro 2 to a new antenna today - it seems to be covering well because I can hear the IDer all over town. 

The new antenna is a dedicated UHF anntenna for Metro 2. Enjoy!

closed 2023-09-23 by mgenelins