mgenelins/w0yc-repeaters #30: NXDN system move from STP campus and reprogram

mgenelins, 2023-09-23:
From Issue #14 

Now that Jason has the Kenwood Nextedge capable repeater, I believe the TODOs are:

- [ ] Reprogram for new frequency. Perhaps 146.940 or 145.230. 
- [ ] Find a new duplexor for this system. VHF.
- [ ] Program one channel for High power FM
- [ ] Program One channel for Low poer FM
- [ ] Program one channel for High Power Next Edge
- [ ] Program One channel for low Power next edge
- [ ] Put on the air at Jason's house for awhile to burn in
- [ ] Find a site for this system and place on the air.
- [ ] Get Jason and Matt and club some nextedge VHF user radios for cheap.

Jason feel free to modify this list. This is just me guessing.

mgenelins, 2024-06-02:
Hey @Deathn00dle any updates on that Kenwood VHF system?

Deathn00dle, 2024-06-02:
@mgenelins we are on the obtain duplexer step. Thoughts on getting an MMDVM for this one and making it an all mode machine?

mgenelins, 2024-06-02:
Sure! Let me know what to purchase, or I can reimbuse you. 

Would all mode include Next-Edge?