mgenelins/w0yc-repeaters #35: Shack Roof

BendixKingHam, 2024-01-27:
Fix shack roof leak so snow melt doesn't keep dripping in with heat on.

mgenelins, 2024-10-18:
Dick and Matt worked on round one back in August/September.
Sean and Dick did round two of roof patching in September/October.

Still has minor leaks, even after repainting the hamshack ceiling.

mgenelins, 2024-10-18:
Sean noticed poor workmanship on the west side of the roof. The flashing was barely nailed to the roof, which is one likely source of leaks.

- [ ] Consider purchasing FlexSeal prodct to find leaks along west trim piece.

mgenelins, 2024-11-09:
Small leak still exists, now is located over newly installed dispatch console desk.

mgenelins, 2024-11-09:
- [ ] Jason suggested to Ian that we install a gutter to redirect small leak away from desk at the ceiling level.