mgenelins/w0yc-repeaters #39: Metro 4: St. Paul site visit to run tests on performance.

mgenelins, 2024-06-04:
Metro 4 is performing better then before new antenna install, not to system specs.

- [ ] Notify mygmrs.com people that a new antenna has been installed and we are ready for coverage testing.
- [ ] Program family PL tone from Kevin request from two months ago.
- [ ] Notify Kevin that his family PL tone is activated.
- [ ] Bring watt meter and re-check SWR level on Metro 4.
- [ ] Run NanoVNA through duplexor on Metro 4. Was putting out 13 watts
- [ ] Repogram RF decks on Metro 4 Transmitter to higher power to make up for 13 watts previously measured.

mgenelins, 2024-10-18:
After further listening, it's the Metro 4 Motorola GP300 UHF receiver that is simply off frequency. Users are barely getting in because it needs a tuneup/replacement.

mgenelins, 2024-10-18:
6/2024: New antenna installed, looking for coverage reports!
is up on 
https://mygmrs.com/repeater/3856

[image]

KD0TLS, 2024-10-18:
So, where does this leave Metro 16?

mgenelins, 2024-10-24:
@KD0TLS Same place as before; sitting in my garage at the moment.

KD0TLS, 2024-10-24:
Uhh... I mean it sounds like it's not going up at the St. Paul campus like
we planned.

mgenelins, 2024-10-29:
@KD0TLS What would make you say that? Plans have not changed here, just updating the Metro 4 issue to clarify what's happening with this system.

mgenelins, 2024-11-09:
@KD0TLS I updated the Title for this issue to reflect more clarity around our approach. Does this help?

mgenelins, 2024-11-10:
- [x ] Metro 16 on the air in Como Park now in testing phase. Low coverage site, testing.