mgenelins/w0yc-repeaters #40: Metro 2 RX

KD0TLS, 2024-06-29:
The RX on the Metro 2 GMRS repeater has declined significantly. This repeater always has transmitted better than it received. I suspect antenna issues, as little on the repeater end could change and allow the repeater to still function. I would suggest an antenna analyzer be used to determine if the current antenna/feedline is still adequate. 
 I have been able to access this repeater reliably with a 5w station in Plymouth since it first went up. That is no longer the case. A 25w mobile station in Minneapolis proper also now fails to ker-chunk the repeater.
 
Todd D KD0TLS

KD0TLS, 2024-07-22:
The Metro 2 GMRS repeater (462.675) has been serviced. The PSU failed. 
Per Action Radio, these TKR-850 repeaters require a slight re-tuning every two years or so, and this was also done.
Wherever and whenever it goes back on-air, a crude wattmeter should be used to determine if it is set to full power or low power, which is controlled by a small unmarked push-button in the lower corner of the front panel. There is no way to know by simply looking at it. 
 
Todd D.

mgenelins, 2024-07-23:
> ID: <mgenelins/w0yc-repeaters/issues/40/2243949769 @ github. com>

Todd,

Awesome news! Thanks for taking Metro 2 into the professionals!

Yours,

-Matt

-----
Regards,
---Matt Genelin---
Vice President
Gopher Amateur Radio Club at The University of Minnesota ( https://w0yc.org/ )
Sr. Solutions Architect, DevOps Engineer
ARS N0YNT/W0USD, GMRS WQCQ790
Cell: +1 (612) 605-5382 Texts or call.
NNNNN

mgenelins, 2024-10-18:
- [ ] Replace the antenna

mgenelins, 2024-11-09:
We need to identify which antenna Metro 2 is presently on. That is still unknown, as it's a borroed / abdondoned antenna on the site.